AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1263: Interrupted Calibration - Unclear Dialog Option

celendis, 2024-07-06:
**Is your feature request related to a problem? Please describe.**
When battery calibration is interrupted for any reason, an alert is presented with the message: "Calibration Mode has been interrupted. Do you want to continue the calibration?" The options provided are "Cancel" and "OK."

**Ambiguity in Options**: The options "Cancel" and "OK" are unclear. Users may wonder if "OK" means accepting that the calibration has stopped or if it means continuing the calibration process. This ambiguity can cause confusion and hinder the user's decision-making process.

**Lack of Preferred Option Indication**: Neither option is highlighted or indicated as the preferred choice, leaving users without guidance on the recommended action.

**Describe the solution you'd like**
**Rename the "OK" Option**: Change the label of the "OK" button to "Continue." This will make it immediately clear to users that selecting this option will resume the calibration process.

**Preferred Option Highlight**: Highlight the "Continue" option to indicate it as the recommended action, providing additional guidance to users.

**Describe alternatives you've considered**
Given the specific nature of this issue, renaming the "OK" option to "Continue" and highlighting it as the preferred action seemed the most straightforward solution.

**Additional context**
Here is the alert from  AlDente Pro 1.27 
[image]

MatthiasKerbl, 2024-07-15:
Hi @celendis,

Thank you for your feedback. We are considering it and will make improvements with the next update.

Best,
Matthias

MatthiasKerbl, 2025-04-15:
Hi @celendis,

We have updated the dialog options now in the latest version.

Best,
Matthias

closed 2025-04-15 by MatthiasKerbl