AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1272: Unable to install helper

kuyvub, 2024-07-15:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.
[image]
Unable to install helper, My OS is Big sur

MatthiasKerbl, 2024-07-15:
Hi @kuyvub,

Please fill out the questions above so we can better assess your issue.

Best,
Matthias

kuyvub, 2024-07-15:
Thank you for getting back to me so quickly.
[image]
After entering the password and installing the auxiliary program, it shows that it is successful but in fact it is not successfully installed, it does not work, the pop-up interface is my screenshot like this, the previous old version did not have this problem, and it cannot be used after updating to the latest version
[image]

kuyvub, 2024-07-15:
[image]
Big sur is not supported now?Is more setup is required?Please tell me because this software is very useful

kuyvub, 2024-07-20:
Back to the old version，Hope you fix this problem in the future

closed 2024-07-20 by kuyvub