AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1274: AlDente causing high CPU usage?

justusrk, 2024-07-18:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBookPro11,5, (A1398), Intel, Quad i7
 - macOS Version (https://support.apple.com/en-us/HT201260): 12.7.5 (21H1222)
 - Charger Model (Apple, Third Party, Wattage,...): Apple 85W Magsafe 2 (original adapter that came with the device, did repair the magsafe port )
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free: AlDente Pro
 - AlDente Version: 1.27.3
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - 
[image]
[image]
[image]
[image]
[image]

 - External Monitor connected? yes, 24" BenQ 1080p
 - MacBook used in Clamshell Mode (with the lid closed)? no, using the Macbook Screen also for work

**Describe the bug**
I am not sure if this is because of AlDente, but I don't face this issue when AlDente is turned off. I have experimented this a couple of times but I face this only when AlDente is turned on.

When AlDente is turned on, my CPU is overloaded. As crazy it would seem. I suppose a lot of factors are at play but reporting this just in case.

configd and powerd seem to take up high cpu usage. Especially configd. When I turn off AlDente the CPU comes back down.

**To Reproduce**
Steps to reproduce the behavior:
1. Just turn on AlDente on an old Intel MBP running macOS Montrey
2. Use your system for some time
3. Experience High CPU usage and laggy OS. 
4. Check Activity monitor

**Expected Behavior**
Just be able to use my system normally.

**Additional context**
I have replaced the MagSafe 2 port because the original one just died.
I have replaced my battery with an iFixit battery because the original one just died.

Sorry if this will be a waste of time but reporting just in case.

justusrk, 2024-07-19:
I saw this happen even with AlDente turned off. Apparently it seems to be caused by non-OEM batteries. 

Only potential solution I found is to use OpenCore Legacy Patcher and disabling firmware throttling.
(https://forums.macrumors.com/threads/macbookpro-mid-2015-mds_stores-or-configd-cpu-high-load.2380126/page-5?post=32601823#post-32601823)

Wondering if there is anything AlDente can do to help users with non-OEM batteries. 🤔

nowhitespace, 2025-06-02:
Hi,
**I can confirm this bug and it occurs also with original batteries.**

I just bought an AlDente Pro license for my MacBook Pro (2017) that also has an Intel i7 CPU (dual core, quad threads) and when I have Al Dente's Dashboard open it uses 25-35% CPU constantly. Activity Monitor shows that Al Dente is the culprit.

If I close Al Dente's dashboard (Al Dente is still running in the background), cpu utilization drops instantly back to normal (less than 10%).

I am shattered that Al Dente's dashboard consumes apparently a crazy amount of CPU ressources on Intel macbook pros.

It would be awesome to improve this as I actually find the dashboard useful.

Thanks, HAnnes

MatthiasKerbl, 2025-06-03:
> Hi, **I can confirm this bug and it occurs also with original batteries.**
>
> I just bought an AlDente Pro license for my MacBook Pro (2017) that also has an Intel i7 CPU (dual core, quad threads) and [...]

Hi @nowhitespace,

Thank you for your feedback. The next update of AlDente will bring down the CPU usage of the dashboard significantly when it is not in focus.

Best,
Matthias

nowhitespace, 2025-06-06:
Great news! Thank you!

MatthiasKerbl, 2025-06-28:
Hi everyone,

We have just released AlDente 1.34, which includes a fix for this issue. To get AlDente 1.34, you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2025-06-28 by MatthiasKerbl