AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1291: Implemente a way to revoke all activations

gesperon, 2024-08-11:
**Is your feature request related to a problem? Please describe.**
As we are human, sometimes we forget to deactivate the license on machines before we wipe them, sell them, etc, Occupying  license activations which aren't really being used.

**Describe the solution you'd like**
A way for an authenticated client - email and license should be enough - to disable previous activations. Maybe all activations, pick from a list, or simply the oldest one.

**Describe alternatives you've considered**
I tried contacting support, however it goes thru paddle and from there haven't heard anything. Regardless, it is inconvenient for such a simple operation not to be automated. I just issued a refund request although I plan to make another purchase just to get around this problem.

MatthiasKerbl, 2024-08-11:
Hi @gesperon,

This is definitely something we want to implement.
Right now, you can text our support and we will deactivate your license within a few days. You can find our support email address in your purchasing emails.

Best,
Matthias

gesperon, 2024-08-11:
Hi @MatthiasKerbl thanks for the quick reply. I shoot an email around two days ago, but got a response from paddle. I also tried the email from the receipt, but haven't anything back in either case. I feel it can get confusing if support works on the issue at the same time paddle process a refund request, so maybe it's better at this point just to refund/rebuy (lucky for me I was still in the return window).