AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1296: AlDente dies sometimes (bug?) - restart it automatically.

kapitainsky, 2024-08-17:
I have noticed that sometimes AlDente dies when my laptop is asleep and as a result battery is fully charged.

Whatever reason of such behaviour I think the simple countermeasure could limit its impact by simply starting AlDente again - launchd job checking if AlDente is running and if not starting it again would solve the issue. Ideally such behaviour should be configurable - "keep alive" option in UI for example.

MatthiasKerbl, 2024-08-17:
> I have noticed that sometimes AlDente dies when my laptop is asleep and as a result battery is fully charged.
>
> Whatever reason of such behaviour I think the simple countermeasure could limit its [...]

Hi @kapitainsky,

To assess your issue better, please share the crash reports of AlDente. You can find the crash reports by opening the “Console” app and navigating to “Crash Reports” on the left side. Afterward, you will see a list of all crash reports. Now, please export all AlDente crash reports individually by right-clicking on them and saving them as a .ips file.

Best,
Matthias

kapitainsky, 2024-08-17:
It is great that AlDente saves it! Here you are:

[AlDente-2024-08-10-051312.ips.zip](https://github.com/user-attachments/files/16645096/AlDente-2024-08-10-051312.ips.zip)
[AlDente-2024-08-17-050254.ips.zip](https://github.com/user-attachments/files/16645098/AlDente-2024-08-17-050254.ips.zip)

Please note that whatever reason there can be always some new bug in the future. "Keep alive" would provide some solution - even if cumbersome.

MatthiasKerbl, 2024-08-19:
Hi @kapitainsky,

Thank you for the crash reports. They helped us a lot to analyze the issue and we will try to fix it with the next update.
Also, regarding your "keep alive" suggestion, it sounds interesting and we will look into it.

Best,
Matthias

kapitainsky, 2024-08-19:
Thank you for looking into it.

MatthiasKerbl, 2026-10-07:
Hi @kapitainsky,

Thank you again for the crash reports. Since your report, we have made many stability and memory improvements, especially when AlDente is running for a long time.

Regarding your "keep alive" suggestion: if you want to make sure that your MacBook does not charge to 100% when AlDente is not running, you can enable "Stop charging when app closed" within AlDente at Sleep Behavior/Show Advanced Settings.

I will close this issue for now. If AlDente still quits unexpectedly with the latest version, just let me know and I can easily reopen it.

Best,
Matthias

closed 2026-10-07 by MatthiasKerbl