AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1299: Control MagSafe LED On/Off from popover menu

benjvfr, 2024-08-18:
Hello,

It could be great to add an option to control the On/Off MagSafe LED from the popover menu.

Use case : when you use the Mac at night / evening, the green LED can be quite annoying : adding an option to disable / enable the MagSafe LED in the popover menu for quick access could be really useful in my opinion.

MatthiasKerbl, 2024-09-03:
Hi @benjvfr,

We have just released AlDente 1.28.1 which adds a setting to automatically turn off the MagSafe LED if the MacBook is asleep. To get AlDente 1.28.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

MatthiasKerbl, 2024-09-03:
> Hi !
> 
> Just updated to v1.28.1, appreciates the new design ! And yes, the feature "Turn off the MagSafe LED while Mac is sleeping" is working.
> 
> But it's not really the behavior explained in this issue : here the idea is to use the MacBook and have the ability to turn on/off the MagSafe LED from the Popover Menu (or have a shortcut / quick access to disable / enable quickly).

Hi @benjvfr,

Glad to hear that you like the new design.
Since the new setting solves the mentioned use case I thought the quick control setting of the MagSafe LED is obsolete. However, we will think about implementing it in the future.

Best,
Matthias

benjvfr, 2024-09-03:
Thank you for considering this feature in the future.

Actual settings are great redesign, but to disable / enable the LED, I still need 4 clicks (icon battery -> settings -> appearance -> LED setting selection). If an icon / button is displayed in the popover menu, I only need 2 clicks.

MatthiasKerbl, 2024-11-10:
> > > Hi !
> > > Just updated to v1.28.1, appreciates the new design ! And yes, the feature "Turn off the MagSafe LED while Mac is sleeping" is working.
> > > But it's not really the behavior explained in this issue : here the idea is to use the MacBook and have the ability to turn on/off the MagSafe LED from the Popover Menu (or have a shortcut / quick access to disable / enable quickly).
> > 
> > 
> > Hi @benjvfr,
> > Glad to hear that you like the new design. Since the new setting solves the mentioned use case I thought the quick control setting of the MagSafe LED is obsolete. However, we will think about implementing it in the future.
> > Best, Matthias
> 
> @MatthiasKerbl I am using MacBook pro 14 inch with M4 chip and the version of AlDente pro is v1.28.6. The LED was turned off, when the MacBook is sleeping, the LED turns on automatically. Please check it out.

Hi @iminfine,

I am not sure if I understand your issue correctly. Could you explain it in more detail?

Best,
Matthias

MatthiasKerbl, 2024-11-13:
Hi @iminfine,

If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias