AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1301: Disable Sleep until Charge Limit feature when closing the lid

melody-cheung520, 2024-08-22:
With the recent update, I saw now the screen will be turned off if the lid was closed. But, I want this feature respect the option I've set in the settings app. 1 min after locking the screen. But "Disable Sleep until Charge Limit" didn't let the system know the display is being turned off and starts the 1 min countdown. I tried to enable the "Lock Screen when Closed" function, but this option instantly locks my screen whenever I closed the lid and went for a coffee break.

[image]

MatthiasKerbl, 2024-08-27:
Hi @melody-cheung520,

We will try to look into this and take the macOS setting into consideration.

Best,
Matthias

MatthiasKerbl, 2025-05-15:
Hi @melody-cheung520,

We have just released AlDente 1.33, which includes a fix for this issue. To get AlDente 1.33, you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias