AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1302: Macbook not charging past 80%

T-Tonic, 2024-08-29:
**MacBook:**
 - M1 MacBook Pro 14 inch
 - macOS Sonoma 14.6.1
 - AlDente Pro 1.27.3

**Describe the bug**
My MacBook won't charge past 80% no matter what I set as the charge limit. 

**To Reproduce**
The only thing I could think of that could have caused this is that I started a calibration last week and interrupted it part way through as I had to go on a trip.

**Expected Behavior**
I would expect the MacBook to charge to a 100% if I set the limit to 100% or perform a calibration.

**Screenshots and Screen Recordings**
[image]
[image]

**Additional context**
I usually keep my charge limit at 80% and only charge to 100% when doing a calibration or when going on the occasionally longer trip.

I have tried the following things all multiple times:
- Uninstalling and reinstalling the helper application
- Uninstalling and reinstalling AlDente
- Shutting down the MacBook, disconnecting everything, and closing the lid for several minutes
- Using different cables and chargers
- Discharging and recharging the battery

MatthiasKerbl, 2024-08-29:
Hi @T-Tonic,

I am sorry to hear that you are having issues with AlDente. To assess your issue better, please share a debug file. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

T-Tonic, 2024-08-29:
Here is the debug file:

[aldentepro-debug-2024-08-29-14:03:15.pasta.txt](https://github.com/user-attachments/files/16798686/aldentepro-debug-2024-08-29-14.03.15.pasta.txt)

MatthiasKerbl, 2024-08-29:
> Here is the debug file:
> 
> [aldentepro-debug-2024-08-29-14:03:15.pasta.txt](https://github.com/user-attachments/files/16798686/aldentepro-debug-2024-08-29-14.03.15.pasta.txt)

Hi @T-Tonic,

It seems like charging is inhibited by a mechanism AlDente does not use. This is either the case because you have Apple's Optimized Battery Charging enabled or you have used or you are still using different charge limiter apps. It is recommended to check if you have Optimized Battery Charging enabled. If yes, disable it. Otherwise, if you have other charge limiter apps installed, set the charge limit to 100% with them and uninstall them again.

Best,
Matthias

T-Tonic, 2024-08-29:
Optimised Battery Charging is disabled. I am not using any other charge limiter apps (and have never used any).

T-Tonic, 2024-09-01:
Is it possible to fully uninstall the app? Even after uninstalling the Helper tool first, followed by AlDente itself, the app’s settings persist after I reinstall it.

Honestly not sure what else to try at this point...

MatthiasKerbl, 2024-09-02:
Hi @T-Tonic,

Thank you for your updates. We don't know why your system's charge limit got stuck without using Optimized Battery Charging or other charge limiter apps. This is especially strange since AlDente has never used this mechanism to limit charging. Therefore, there is no indication that AlDente caused your issue. However, we have implemented a background safety feature that will reset the mechanism used in your case. This should solve your issue. The new update will be released this week.

> Is it possible to fully uninstall the app?

Yes, if you follow the steps in our [Uninstallation Guide](https://apphousekitchen.com/how-to-uninstall-aldente-pro/), all traces of AlDente are gone from your system.

Best,
Matthias

MatthiasKerbl, 2024-09-03:
Hi @T-Tonic,

We have just released AlDente 1.28.1 which includes a fix for this issue. To get AlDente 1.28.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2024-09-03 by MatthiasKerbl

T-Tonic, 2024-09-03:
I just updated to 1.28.1 and it fixed the issue. Thank you very much!