AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1304: Need to install helper after every reboot on aldente pro

marcjuhl, 2024-08-30:
**MacBook**
 - MacBook Model (2021 m1pro macbook pro 14"):
 - macOS Version (15.1):
 - Charger Model (official 67w charger that came with it):
 - Charging Cable (usb c and magsafe 3):
 - AlDente Pro
 - AlDente Version: 1.27.3
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - 
[image]
[image]
[image]
[image]
[image]
[image]

 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 
 - External Monitor connected?
 no
 - MacBook used in Clamshell Mode (with the lid closed)?
 no

**Describe the bug**
A clear and concise description of what the bug is.
Every time i restart my mac it asks for me to install the helper application, no matter if i already installed it..

**To Reproduce**
Steps to reproduce the behavior:
1. Restart your mac
2. click the aldente menu
3. it says i need to install the helper app even though i have

**Screenshots and Screen Recordings**
[image]

marcjuhl, 2024-08-30:
Sorry heres the debug file 
https://drive.google.com/file/d/14FRUlTR7K6ra_f-NP5-lsfnR3OsWQr-1/view?usp=drive_link

b159732000, 2024-09-02:
Me too on M1 Max, OS 14.6.1（23G93）

MatthiasKerbl, 2024-09-03:
Hi everyone,

Usually, problems regarding the helper installation can be fixed with a simple restart of your MacBook.
If that does not work for you, please follow these steps:

- Please go to Finder navigate to /Library/PrivilegedHelperTools/ and delete the file: "com.apphousekitchen.aldente-pro.helper"
- Then, please navigate to Library/LaunchDaemons/ and remove "com.apphousekitchen.aldente-pro.helper.plist" 
- Also, remove your AlDente application and reboot your MacBook.
- Then, install it as if it was freshly downloaded.

At last, once the new helper is installed again, please check if AlDente is allowed to run in the background in the "Login Items" section of the macOS settings. This could also be the culprit.

Best,
Matthias

closed 2024-09-03 by marcjuhl

marcjuhl, 2024-09-03:
That worked thanks!

inspiredearth, 2025-04-10:
Same issue. Been occurring for months. Currently on macOS 15.3.1, on M1 platform.
Will try above mentioned manual deletion of helper file, etc.