AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1309: Battery charing to 100% with Aldent on MacBook M3 Max

duffysk, 2024-09-03:
Hello,

I'm using Aldente 1.28.1 on MacBook Pro with M3 Max setup. While it worked well on old Intel silicon it does not on the new one running on Sonoma 14.6.1 (23G93).
I have Aldente charging limit set to 70% but battery keeps charging to 100%.

MatthiasKerbl, 2024-09-03:
Hi @duffysk,

Please fill out the following questions:

**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

duffysk, 2024-09-04:
MacBook (please complete the following information):

MacBook Model ([https://support.apple.com/en-us/HT201608 ](https://support.apple.com/en-us/HT201608)): MacBook Pro, 14-inch, Nov 2023, model number: Z1AW0018RN/A
macOS Version ([https://support.apple.com/en-us/HT201260 ](https://support.apple.com/en-us/HT201260)): 14.6.1 (23G93)
Charger Model (Apple, Third Party, Wattage,...): Apple
Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe 3
AlDente Pro or AlDente Free: AlDente Free
AlDente Version: 1.28.1
List or add screenshots of all enabled/disabled settings in AlDente: 
[image]
[image]

Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog( [https://apphousekitchen.com /how-to-generate-and-share-a-debug-file/ ](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/)).

[aldentepro-debug-2024-09-04-11:01:30.pasta.txt](https://github.com/user-attachments/files/16864067/aldentepro-debug-2024-09-04-11.01.30.pasta.txt)

External Monitor connected? Yes

MacBook used in Clamshell Mode (with the lid closed)? No

Describe the bug
A clear and concise description of what the bug is.
Even when AlDente charging cap is set to 70%, MacBook keeps battery charged to 100%

To Reproduce
Steps to reproduce the behavior:

Go to '...'
Click on '....'
Scroll down to '....'
See error
- No error visible
Expected Behavior
- Battery to be charged to the cap as set in AlDente app

A clear and concise description of what you expected to happen.
- I'd like battery to be charged to the level set by limit in AlDente

Screenshots and Screen Recordings
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

Additional context
Add any other context about the problem here.

PerryFinn, 2024-11-21:
I encountered the same issue on my MacBook Pro M3 max. Has there been any progress on this issue?😁

MatthiasKerbl, 2024-11-21:
Hi @PerryFinn,

I am sorry to hear you are experiencing this issue.
If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

PerryFinn, 2024-11-21:
Alright! I'll reinstall the software when I have time and share the debug files.

PerryFinn, 2025-02-06:
After using the latest version, the software is working normally, that's so cool!

https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/releases/tag/1.30

MatthiasKerbl, 2025-02-06:
Hi @duffysk,

Have you tried our newest version of AlDente and do you still experience this issue?

Best,
Matthias

duffysk, 2025-02-06:
@MatthiasKerbl  seems like it's working fine now.

Thanks!

Best regards,

Rado

closed 2025-02-06 by MatthiasKerbl