AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1310: Losing functions of Remove Icon on Dock

JarrettChen217, 2024-09-04:
Recently I noticed that AlDente's icon had changed, so I went to adjust it, but the icon on the dock was out, and according to the FAQ I couldn't find a way to turn it off in Appearance. Also I noticed that some of the things in the Other section have disappeared, so what can I do about it?
[image]

MatthiasKerbl, 2024-09-04:
Hi @JarrettChen217,

Thank you for reporting this bug. It will be fixed soon.

Best,
Matthias

MatthiasKerbl, 2024-11-21:
Hi @JarrettChen217 ,

We have just released AlDente 1.29, which includes a fix for this issue. To get AlDente 1.29 you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Best,
Matthias

closed 2024-11-21 by MatthiasKerbl