AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1312: Can't stop calibration when charger unplugged

mickgotwings, 2024-09-05:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBookPro18,4 M1 Max
 - macOS Version (https://support.apple.com/en-us/HT201260): Sonoma 14.5
 - Charger Model (Apple, Third Party, Wattage,...): N/A
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): N/A
 - AlDente Pro or AlDente Free: AlDente Pro
 - AlDente Version: 1.28.1
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
When the charger is unplugged, it is impossible to stop the calibration after it has already started

**To Reproduce**
Steps to reproduce the behavior:
1. Start calibration
2. Unplug the charger
3. Try to stop the calibration
4. See error "Calibration Mode is only available when the charger is connected."

**Expected Behavior**
Calibration stops, even when not on charger

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.
[image]

**Additional context**
Tested during both planned and manual calibration.

MatthiasKerbl, 2024-09-09:
Hi @mickgotwings,

Thank you for your detailed bug report. We will try to fix it with the next updates.

Best,
Matthias

mickgotwings, 2024-09-19:
I can see it has been resolved in 1.28.3.
Can confirm it works flawlessly at my end.
When charger is unplugged, I can stop the calibration, but cannot start one, which is just what I expected it to be.
Thank you for that! I believe the issue can be closed.

closed 2024-09-19 by MatthiasKerbl