AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1314: Add an option to switch to AIDente old icon

zzhou98, 2024-09-05:
Thank you for your continued updates and maintenance. 

The app icon has been updated since version 1.28, and it looks great! 

But may I ask could you please add an option to switch between the new and old icons? I think the old version of the menu bar icon is great (sorry the new version is a bit too thin)

Many thanks

MZ-spec73, 2024-09-07:
Yeah, I absolutely agree, the old logo is definitely better

MatthiasKerbl, 2025-04-23:
Hi everyone,

Thank you for your feedback.
We think that our current offering of five different menubar icons should be enough for most users. However, if the demand for the old icon is big enough, we are also open to bringing it back as an option.

Best,
Matthias