AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1315: Display both macOS and hardware battery percentage in the menu bar.

devMEremenko, 2024-09-05:
Hi ALDente team,

**Describe the solution you'd like**
Display both macOS and hardware battery percentage in the menu bar.

**Is your feature request related to a problem? Please describe.**
Nope, love your app 🍝 ❤️ 

Max

sebmets, 2024-11-05:
+1. This is the only reason I still keep my MacOS battery indicator in the menu bar. Otherwise I'd remove it altogether.

MatthiasKerbl, 2026-10-06:
Hi @devMEremenko,

We have just released AlDente 1.39, which makes the hardware battery percentage available in the stats widget and the menu bar items catalog. This way, you can display both the macOS and the hardware battery percentage at the same time. To update to the latest version, open the Dashboard, click on the gear icon in the bottom-left corner, and click on Check for Updates.

I will close this issue for now. Please report back if this works for you.

Best,
Matthias

closed 2026-10-06 by MatthiasKerbl