AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1316: Popover UI has no colors

Tunanika, 2024-09-07:
**MacBook:**
 - MacBook Model: Macbook Air M1 2020 8gb
 - macOS Version: Sonoma 14.6.1
 - Charger Model: 65W Third Party
 - Charging Cable: USB-C
 - AlDente Pro or AlDente Free: AlDente Free
 - AlDente Version: 1.28.2
 - List or add screenshots of all enabled/disabled settings in AlDente:
[image]

 - External Monitor connected: Yes/No (Does not affect bug)
 - MacBook used in Clamshell Mode: No

**Describe the bug**
The new UI is colorless and transparent making it hard to see and use. I am unsure if this is intentional or just an issue on specific models.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Aldente Version 1.28.1 or higher
2. Click on the menu bar icon to open the popover window
3. See the colorless UI

**Expected Behavior**
Old UI with blue slider/buttons

**Screenshots and Screen Recordings**
[image]

Tunanika, 2024-09-22:
Issue persists with MacOS Sequoia (15.0)

MatthiasKerbl, 2026-10-06:
Hi @Tunanika,

Since this report, we have completely redesigned the popover and the whole app appearance with AlDente 1.36. Additionally, there is now a "Reduce Transparency" setting for everyone who prefers the older-style materials.

I will close this issue for now. If you are still experiencing this issue with the latest version, just let me know, and I will reopen it.

Best,
Matthias

closed 2026-10-06 by MatthiasKerbl