AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1319: Add animations to the power flows

devMEremenko, 2024-09-12:
[Enhancement]

Hi!
I would suggest animating the power flows to better the app's perception.
Currently, the middle section instantly "jumps" (expands and collapses) when the app receives new values.

[image]

MatthiasKerbl, 2025-04-23:
Hi @devMEremenko,

I am not sure I understand your recommendation completely.
Do you mean a smoother transition when the values change?

Best,
Matthias

devMEremenko, 2025-04-25:
Hey @MatthiasKerbl,
That's correct. It would be great to animate value changes using a spring animation.

https://github.com/user-attachments/assets/364c74f6-9cdd-45f9-822e-76f684d01bc4