AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1323: Calibration not 100% hold but 98.98%

Volodymyr-13, 2024-09-17:
After calibration gets into step Hold 100% its not holding 100%, its less: 

[image]

1.28.2

MatthiasKerbl, 2024-09-17:
Hi @Volodymyr-13,

Within the one-hour time window, AlDente will charge the MacBook to 100%. Afterward, it will start discharging again.

Best,
Matthias

Volodymyr-13, 2024-09-17:
I monitored this for this 1 hours and it was 99% all the time, then started discharge. 

I think it's time to not round % of charge from raw values(I'm talking about your ignorance of the bug with 79.49%).

MatthiasKerbl, 2024-09-17:
Hi @Volodymyr-13,

Thank you for your feedback. Depending on the calibration of the MacBook, some don't go to 100% (Hardware Battery Percentage) at all. The important thing is to let the charging power go to 0 and this always happens within the one hour time window from our experience.

Best,
Matthias

closed 2025-04-15 by MatthiasKerbl