AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1326: Chargin limit doesn't work since I updated to MacOS Sequoia

yoav28, 2024-09-17:
**MacBook (please complete the following information):** Macbook air m2
 - macOS Version (https://support.apple.com/en-us/HT201260): MacOS Sequoia
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe
 - AlDente Pro or AlDente Free: Free
 - AlDente Version: 1.27.3

**Describe the bug**
A clear and concise description of what the bug is.
Aldete chargin limit does'nt since I updated to MacOS Sequoia yesterday, I set the limit to 80% but it charges to 100% anyways.
Discharge works as usual so the issue is only with limitation

[image]

MatthiasKerbl, 2024-09-17:
Hi @yoav28,

Unfortunately, AlDente temporarily does not work after a macOS update for some users.

To get AlDente to work again, please try the following steps which have worked for many users before:
1.  Unplug and discharge your MacBook to less than your current charge limit in AlDente
2. Turn off your MacBook and close the lid for a couple of minutes
3. Unplug the power brick from the outlet and wait a couple of minutes
4. Turn on your MacBook and plug it in again
5. Check if it stops charging at the charge limit set in AlDente

We are aware of this bug for a while now but we’re so far not able to fix it. However, we will keep trying.

Cheers,
Matthias

MasterOfMouse, 2024-09-18:
I think the devs need a couple of days to make AlDente compatible with the newest version of macOS. 

I updated to macOS 12 before checking if AlDente was ready. Since then I have always waited till AlDente is ready.

MatthiasKerbl, 2024-09-30:
> I think the devs need a couple of days to make AlDente compatible with the newest version of macOS.

Hi @MasterOfMouse,

macOS 15.0 is fully supported by AlDente 1.28.4.
The issue described @yoav28 is caused by macOS updates by some users and can be fixed by following the steps mentioned above.

Best,
Matthias

MatthiasKerbl, 2024-09-30:
> Hi! Are there any timeline updates on this?

Hi @mickgotwings,

macOS 15.0 is fully supported by AlDente 1.28.4.

Best,
Matthias

mickgotwings, 2024-10-01:
Oh, I thought it was a persistent problem, and "temporary" meant "while we fix it".
Thank you for clarifying this!

adit-s, 2024-12-08:
I upgraded to Sequoia yesterday and ran into the same problem. 
In Settings->General->Login Items & Extensions, I set AlDente to be "Allow in Background."
Since then, AlDente seems to be working normally. 
I didn't need to follow the steps outlined above by @MatthiasKerbl .
Hopefully, this info can help others.

1
[image]

2
[image]

ralfbergs, 2025-02-02:
I have the same problem, but I just started using AlDente a week or two ago, and I was already running Sequioa when I installed AlDente. 🤷🏻‍♂️

petiteplanete, 2025-02-07:
Same here: Sequoia 15.3 installed this morning, and now AlDente does not work.

closed 2025-04-15 by MatthiasKerbl