AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1332: Temperature  limiter is semi-broken

Sidepie, 2024-09-18:
**MacBook (please complete the following information):**
 - M1Pro
 - Sequoia
 - Apple 120W
 - USB-C, Apple standard
 - AlDente Pro
 - AlDente Version: 1.28.3
 - External Monitor connected? - No
 - MacBook used in Clamshell Mode (with the lid closed)? - No

**Describe the bug**
On my machine, charging stops when the temperature exceeds 35°C. 
However, once charging stops, it does not restart, even if the temperature falls back below the threshold and more than 5 minutes have passed. 
In other words, the hysteresis process described in the documentation does not occur, except for stopping when the temperature limit is initially reached.

**Expected Behavior**
The charging temperature limit should work as described

**Additional context**
If I manually move the charging limit, up or down, in the popup, with the mouse, the charging is instantly resumed.

MatthiasKerbl, 2024-09-19:
Hi @Sidepie,

If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

jkittikun, 2024-09-20:
I got the same problem. Charging does not continue after heat protection triggered.

- MacBook Air M1
- Sequoia
- Apple 30W
- USB-C, Apple standard
- AlDente Pro
- AlDente Version: 1.28.3

[image]

MatthiasKerbl, 2024-09-20:
Hi everyone,

We have just released AlDente 1.28.4 which includes a fix for this issue. To get AlDente 1.28.4 you need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

Sidepie, 2024-09-20:
Hi,
Thank you for your fast support, everything seems to be in order now.

closed 2024-09-20 by MatthiasKerbl