AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1334: Macbook charging starts and stops every 2-3 seconds

Danzikks, 2024-09-19:
- MacBook Model: M1 pro 16" 2021
 - macOS Version: Venture and Sequoia
 - Charger Model: Apple
 - Charging Cable: MagSafe 3
 - AlDente Pro
 - AlDente Version: 1.27 - 1.28.3
 - External Monitor connected? no
 - MacBook used in Clamshell Mode (with the lid closed)? no

Hi, I've racked my brain what's going on with my charger and can't find the cause of it.
At times my charger starts to turn on and off every 2 seconds and it can be hours, I have to unplug my macbook. I noticed that when this happens, the status bar says a very strange number to charge the battery to full.
Help me please at least understand what the problem is, in the software, in the cable, in the power adapter, in the macbook. I also noticed that if I reboot the macbook, this moment with charging that turns on and off every 3 seconds can disappear.  I've also noticed that unplugging the power adapter can also end it. I also noticed that if you charge the macbook via USB-C, the error goes away, but as soon as you insert MagSafe 3, the error can appear.
This error may not appear for a long time, or it may reappear and disturb me. 

Macbook is new, was purchased 2 months ago. I live in a country where there is no apple service, so all my repairs are at my own expense :(
And I need to find out where the problem is.

Translated with DeepL.com (free version)

[image]

andrewcrook, 2024-09-24:
MacBook Model: M1 MacBook pro 13” 2020
macOS Version:  Sequoia 14.0
Charger Model: Apple
Charging Cable: USB-C
AlDente Pro
AlDente Version: 1.28.4
External Monitor connected? Yes
Monitor: LG 850 4K UK
MacBook used in Clamshell Mode (with the lid closed)? no

Not sure if the "Time to full:” is related to the same issue?
I only just noticed this I get it on a totally different setup and look it’s the **same number is reported** `1092h 15m`
However, I don't have the charging stopping and starting issue.

[image]

Danzikks, 2024-10-01:
The solution to my problem has been found and it is not related to aldente as expected.
The problem was with the Apple 140W adapter

closed 2024-10-01 by Danzikks