AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1336: Lock Screen not dimmed to min brightness

LURSU-MiR, 2024-09-20:
- MacBook Air M3 2024
 - macOS Sequoia
 - Apple Dual USB C 32w Charger
 - MagSafe 3
 - AlDente Pro
 - AlDente Version: 1.28.4
 - List or add screenshots of all enabled/disabled settings in AlDente:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

 - External Monitor connected? - No
 - MacBook used in Clamshell Mode (with the lid closed)? - Yes

**Describe the bug**
When having disable sleep until charge limit on, the screen dims to the lowest possible brightness. But when also activating Lock Screen when closed, the screen does not dim as much as it could when having the lid closed.

**Expected Behavior**
The screen should dim to the lowest possible brightness when having both Disable sleep until charge and Lock Screen when closed on

MatthiasKerbl, 2024-09-20:
Hi @LURSU-MiR,

Unfortunately, I am currently not able to reproduce this issue.
If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

LURSU-MiR, 2024-09-20:
[aldentepro-debug-2024-09-20-19:41:43.txt](https://github.com/user-attachments/files/17079725/aldentepro-debug-2024-09-20-19.41.43.txt)

I have attached a debug file (I had to change its file extension to txt as otherwise it wouldn't have uploaded) 

When I plug the charger in and then close the lid with the setting Lock Screen when closed on, I can see that the screen is not on the lowest brightness setting when opening the Mac barely, so that the sensor does not detect, that the lid is opened, which it is when only disable sleep until charge is enabled.

LURSU-MiR, 2024-09-20:
I have looked at the problem again, but I think, that the display is in fact dimmed correctly, but turns on so quickly, that it is not really visible that it was of in the beginning.

LURSU-MiR, 2024-09-20:
Thank you though for your fast replying!

LURSU-MiR, 2024-09-20:
I found out, that when I open the laptop and lock it manually, it still doesn't sleep, but the display gets turned off completely

MatthiasKerbl, 2025-05-15:
Hi @LURSU-MiR,

We have just released AlDente 1.33, which includes a fix for this issue. To get AlDente 1.33, you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2025-05-15 by MatthiasKerbl