AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1337: Pause Battery charge button

LURSU-MiR, 2024-09-20:
Implement a button, with which you can pause the charging of the battery or even discharge it with a power adapter plugged in at any time, not only when it is at the set limit or above.

MatthiasKerbl, 2024-09-21:
Hi @LURSU-MiR,

We have implemented multiple ways to pause charging, some of which are the following:

- Use the slider bar in the popover window.
- Use the "Limit:" text field to set the charge limit and pause charging.
- Use the "Menubar Right-Click" feature.
- Use the Schedule feature.
- Use the Apple Shortcut implementation.

When it comes to discharging, you also have multiple possibilities:

- Use the Discharge button when the charge limit exceeds the current battery percentage.
- Activate the "Automatic Discharge" feature.
- Use the Schedule feature.
- Use the Apple Shortcut implementation.

I hope that helps.

Best,
Matthias

closed 2024-09-21 by MatthiasKerbl

LURSU-MiR, 2024-09-21:
Hi @MatthiasKerbl,

it would be cool though to have dedicated buttons in the popover menu.

Best wishes,
Lursu