AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1345: Charge page: Aldente disable sleep option 'slider button' is disabled (greyed out) on new M3 MacBook Pro

ravenboy, 2024-10-06:
Aldente disable sleep option 'slider button' is disabled (greyed out) on new M3 MacBook Pro

how can I get it NOT to be greyed out?

MatthiasKerbl, 2024-10-07:
Hi @ravenboy,

Please check out the following article on our blog "[When Does AlDente Disable Sleep on MacBooks?](https://apphousekitchen.com/when-does-aldente-disable-sleep-on-macbooks/)".

Best,
Matthias

ravenboy, 2024-10-07:
thanks, sleep isn't disabled on my Mac.

the page you video doesn't discuss when and why the 'stop charging  sleep 

I see I was unclear in my original post, or perhaps I am missing something.

I want to be able to turn on the 'stop charging when sleeping' control on the charge page in your product. it is never
been enabled so I can slide the control to the ON position

it wasn't enabled when I installed your product
 it didn't enable after the first' calibration' was successfully completed

a click on the 'info' button next to this the  'stop charging when sleeping'  control  takes me to 

Feature Explanation: Stop charging when sleeping
and it doesn't discuss when and why this control is disable/enabled 

it DOES describe what happens when 'Stop charging is activated'

I want to be able to activate it.

MatthiasKerbl, 2024-10-09:
Hi @ravenboy,

Apologies from my side, I misunderstood your issue.
Do you have the feature "Disable Sleep until Charge Limit" enabled?
If yes, this is the reason why "Stop charging when sleeping" is greyed out. This is due to "Stop charging when sleeping" not having any functionality when "Disable Sleep until Charge Limit" is enabled.

Also, when "Stop charging when sleeping" is greyed out and you hover the cursor above the slider, the following text field will appear:
"When activated, charging is automatically
inhibited when the system is entering sleep
mode, even if the desired percentage is not yet
reached. This feature is not available when
Disable Sleep until Charge Limit is enabled."

Best,
Matthias

ravenboy, 2024-10-10:
ahhh I see

you wrote:
Also, when "Stop charging when sleeping" is greyed out and you hover the cursor above the slider, the following text field will appear:

on a new M3 MacBook Pro, and with yversion 1.28.5, hovering the cursor above the slider the text to which you refer DOES NOT appear.  Had that happened, it would have been helpful, and I would not have had to waste your time.

FYI, if text is supposed to appear when hovering over a control elsewhere in your product, on my Mac it doesn't.

this is a new Mac and I can't imagine that I have installed anything that would disable your features. (Sequoia 15.1)

cheers and again, I love this product

MatthiasKerbl, 2024-10-16:
Hi @ravenboy,

Thank you for your feedback, we will look into the popovers.

Best,
Matthias

closed 2024-10-16 by MatthiasKerbl