AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1348: AlDente Pro prevents Mac from sleeping while on battery with lid closed. Probable heat damage

avishaan, 2024-10-09:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): Mac15,10
 - macOS Version (https://support.apple.com/en-us/HT201260): 14.6.1 (23G93)
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Magsafe 3
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.28.4
 - List or add screenshots of all enabled/disabled settings in AlDente:
 
[image]
[image]

 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 
[aldentepro-debug-2024-10-08-18:56:59.pasta.zip](https://github.com/user-attachments/files/17301064/aldentepro-debug-2024-10-08-18.56.59.pasta.zip)

 
 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? Yes

**Describe the bug**
I've had this issue happen a couple times and only realized it was an issue with AlDente when I looked at the bug reports on GitHub. Basically, my laptop was in my bag and when I took it out about 5 hours later, the laptop was scorching hot, 30% battery (originally went in with 90% battery), and about 99F which is in the battery damage level. It feels like I probably did more damage with that one mistake than a year of moderate use.

**To Reproduce**
Steps to reproduce the behavior:
Unplug laptop
close lid
put in bag
~4 hours later
take from bag, laptop will be very hot and almost dead

**Expected Behavior**
I expected it to work like it says in example 2 of the FAQ

> Your MacBook is plugged in, charging and the battery percentage is currently 32% and the Charge Limit is set to 80%. “Disable Sleep until Charge Limit” is activated and you close the lid of your MacBook. You need to use your MacBook on the go after only a couple of minutes, therefore, **you unplug your MacBook although the charge limit is not reached yet. AlDente notices that your MacBook is unplugged and will activate sleep again**. When you start using your MacBook again, the charge level is exactly where it was when the MacBook was unplugged.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

avishaan, 2024-10-09:
x-post on Reddit in case it helps future folks https://www.reddit.com/r/mac/comments/1fzgpar/aldente_pro_prevents_mac_from_sleeping_while_on/

MatthiasKerbl, 2024-10-09:
Hi @avishaan,

I am sorry to hear you are having issues with AlDente, and thank you for the detailed bug report and debug file.
Are you able to reproduce this issue?
If yes, could you generate a debug file directly afterward so that we can analyze what is going on when the bug is happening?

Best,
Matthias

closed 2025-04-15 by MatthiasKerbl