AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1349: Sleep Mode not Working on Mac Book Pro M3 with Docking Station

GregorWedlich, 2024-10-14:
MacBook

	•	MacBook Model: MacBook Pro M3 Max
	•	macOS Version: macOS Sonoma 14.6.1
	•	Charger Model: [FusionDock Max 1](https://ivanky.com/products/ivanky-docking-station-fusiondock-max-1)
	•	AlDente Pro
	•	AlDente Version: 1.28.5
	•	List or add screenshots of all enabled/disabled settings in AlDente: Bitte ergänzen (Screenshots hinzufügen)
	•	External Monitor connected?: Yes, 3 via Ivanky Docking Station 

Describe the bug
When the MacBook Pro is connected to an Ivanky Docking Station and AlDente is active, it does not stay in Sleep Mode. I have tried using the “Stop charging when sleeping” setting, but the issue persists. However, the MacBook stays in Sleep Mode as expected when AlDente is disabled. The Same problem with CalDigit on other Mac.

To Reproduce
Steps to reproduce the behavior:

	1.	Connect MacBook Pro to Ivanky Docking Station.
	2.	Enable AlDente and ensure “Stop charging when sleeping” is active.
	3.	Put the MacBook into Sleep Mode.
	4.	Observe that the MacBook does not stay in Sleep Mode.

Expected Behavior
The MacBook should remain in Sleep Mode when “Stop charging when sleeping” is enabled in AlDente and the Docking Station is connected.

MatthiasKerbl, 2024-10-16:
Hi @GregorWedlich,

Are you using your MacBook in Clamshell mode?
If possible, please share a debug file directly after it happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

GregorWedlich, 2024-10-16:
I dont know how i aktivate Clamshell mode.

The problem exists if the MacBook is connected with a Docking Station (With charging), whether opened or not.

I can send a Debug file, but her in public? :(

MatthiasKerbl, 2024-10-17:
> I dont know how i aktivate Clamshell mode.
> 
> The problem exists if the MacBook is connected with a Docking Station (With charging), whether opened or not.
> 
> I can send a Debug file, but her in public? :(

Hi @GregorWedlich,

Since you are using AlDente Pro, you are eligible for our email [premium support](https://apphousekitchen.com/support/).
Please reference this GitHub issue when you reach out.

Best,
Matthias

closed 2025-07-14 by MatthiasKerbl