AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1352: Problem with clamshell

ThomasDanilo96, 2024-10-18:
**MacBook (please complete the following information):**
 - MacBook Model Macbook air m3
 - macOS Version macOS Sequioia 15.0.1
 - Charger Model Apple
 - Charging Cable MagSafe 3
 - AlDente Pro 
 - AlDente Version: 1.28.5
 - List or add screenshots of all enabled/disabled settings in AlDente:
 
[image]
[image]
[image]

- Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).

[aldentepro-debug-2024-10-18-14:44:16.txt](https://github.com/user-attachments/files/17435639/aldentepro-debug-2024-10-18-14.44.16.txt)

 - External Monitor connected? yes, 2
 - MacBook used in Clamshell Mode (with the lid closed)? yes

**Describe the bug**
Good evening, I have a problem with my MacBook Air M3. I purchased a pro license for AlDente to use it in clamshell mode so that it could automatically discharge and recharge according to settings. Unfortunately, when the charge limit is set, it reaches that limit and maintains the charge at that level without dropping below it by utilizing the sailing mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2024-10-18:
Hi @ThomasDanilo96,

This is the intended behavior of the Sailing Mode and looking at your screenshots, the Sailing Mode is working as intended.
It is a common misconception that the Sailing Mode does actively charge and discharge your MacBook between the sailing range. This would actually be much worse for the battery.
You can read a detailed description of what the Sailing Mode does on our blog "[Feature Explanation: Sailing Mode](https://apphousekitchen.com/feature-explanation-sailing-mode/)"

Best,
Matthias

ThomasDanilo96, 2024-10-18:
Hello @MatthiasKerbl,
Thank you for the prompt response.
Consequently, how should I use AlDente with clamshell mode (I only use it in that mode for work) to protect the battery?

MatthiasKerbl, 2024-10-22:
Hi @ThomasDanilo96,

The Clamshell Mode does not change much regarding how to use AlDente. Choose your preferred charge limit and enable the other features depending on your usage and preference. You can read more about the features on our [blog](https://apphousekitchen.com/blog/).

Best,
Matthias

closed 2024-10-22 by MatthiasKerbl