AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1357: Time to empty instead of Time to full when discharging

kandrei, 2024-10-30:
**Is your feature request related to a problem? Please describe.**
It seems that Aldente is showing "Time to full" even when the battery is discharging. The displayed time is the same one displayed by CoconutBattery, but that one says "Time until empty". Aldente should change the text in this case to "Time to empty"

**Describe the solution you'd like**
If "Battery Current" < 0, change the text to "Time to empty" or "Time until empty"

**Describe alternatives you've considered**
Just comparisons with other software.

**Additional context**
[image]

MatthiasKerbl, 2024-10-30:
Hi @kandrei,

Thank you for your feedback.
Please answer the following questions:

1. Which MacBook model are you using?
2. Which macOS version are you using?
3. Which AlDente version are you using?

Best,
Matthias

MatthiasKerbl, 2024-10-31:
Hi @kandrei,

We have just released AlDente 1.28.6, which includes a fix for this issue. To get AlDente 1.28.6 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

kandrei, 2024-10-31:
Hi @MatthiasKerbl,

I've installed the new version and I can confirm that the fix is working.

Thank you for this quick fix,
Andrei