AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1359: Cannot Update to Newer Version of AlDente Pro 1.28.6

By-pass, 2024-10-30:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): Macbook Pro 16 (Apple M2) 
 - macOS Version (https://support.apple.com/en-us/HT201260): Ventura 13.7 (22H123)
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.28.5
 - External Monitor connected?: No
 - MacBook used in Clamshell Mode (with the lid closed)?: No

**Describe the bug**
The update keeps failing (from 1.28.5, updating to 1.28.6). I have tried restarting Mac and AlDente, but it did not solve the issue. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Setting
2. Click on Update
3. Download the update (1.28.6)
4. Error - Could not successfully update due to an error from unpacking the update files. 

[image]

Thank you for your time and help.

MilesTEG1, 2024-10-30:
Hello,
I've got the same problem (I think, because I don't understand the text in the screenshot posted by @By-pass).

[image]

MilesTEG1, 2024-10-30:
I managed to update by using the `brew upgrade aldente` command.
It removed the old app, and installed the new one.
After reinstalling the Assistant, all good.

amit-bh, 2024-10-31:
This is the error I am getting:

[image]

MatthiasKerbl, 2024-10-31:
Hi everyone,

Thank you very much for reporting this issue. Unfortunately, we cannot reproduce it (yet) but we are looking into it.
If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

MatthiasKerbl, 2024-10-31:
Hi everyone,

We were able to resolve this issue. You should now be able to update AlDente without any issues.

Thank you for your patience.

Best,
Matthias

closed 2024-10-31 by MatthiasKerbl