AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1367: Every time I open AlDente after restart it says "Install Helper"

mmrrttnn, 2024-11-07:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Pro 2020 M1 16GB 13 inch
 - macOS Version (https://support.apple.com/en-us/HT201260): Ventura 13.7.1
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Apple
 - AlDente Pro or AlDente Free: AlDenteFree
 - AlDente Version: 1.28.6
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - 
[image]
[image]
[image]
[image]
[image]
[image]

 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?: No
 - MacBook used in Clamshell Mode (with the lid closed)?: No

**Describe the bug**
A clear and concise description of what the bug is.
After every time the app is being quitted/switched totally off e.g. after a restart it requres helper installation. 

**To Reproduce**
Steps to reproduce the behavior: I do not know if others have the same issue.
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.
I expect not to install "Installer" every time.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

Charlieelliott24, 2024-12-22:
I've been having this same issue since about the same time. Thanks.

MatthiasKerbl, 2025-07-15:
Hi everyone,

Please follow these steps:

1. Please go to Finder and navigate to: /Library/PrivilegedHelperTools/ and delete the file: com.apphousekitchen.aldente-pro.helper
2. Then, please navigate to: Library/LaunchDaemons/ and remove the following: com.apphousekitchen.aldente-pro.helper.plist
3. Also, remove your AlDente application and reboot your MacBook.
4. Then, install as if it were freshly downloaded.
5. Once the new helper is installed again, please check if AlDente is allowed to run in the background in the "Login Items" section of the macOS settings. This could also be the culprit.

Best,
Matthias

kylepetz, 2025-07-17:
This worked!