AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1370: MacBook Pro M4 crashes

thirteenkai, 2024-11-12:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2024-11-12:
Hi @thirteenkai,

Thank you for your feedback.
Unfortunately, M4 MacBooks are not (yet) fully supported by AlDente. I will give updates about this here.
It would be helpful if you could answer the questions above for now.

Best,
Matthias

thirteenkai, 2024-11-12:
MacBook 型号: MacBook Pro M4 Pro
macOS 版本: 15.1 (24B2083)
充电器型号: Apple
充电线:MagSafe 3
AlDente Pro 或 AlDente Free：AlDente Pr
AlDente 版本：1.28.6
列出或添加 AlDente 中所有启用/禁用设置的屏幕截图：
下次发生这种情况时，请分享一个调试文件，以便我们更好地评估此问题。您可以在我们的博客上找到有关如何生成和共享调试文件的指南（[[https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/）。](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/%EF%BC%89%E3%80%82)](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/%EF%BC%89%E3%80%82)
外接显示器已连接？:yes
MacBook 是否以翻盖模式（盖子关闭）使用？:盖子关闭
描述错误：直接闪退

MatthiasKerbl, 2024-11-13:
Hi @thirteenkai,

Could you share a debug file an crash report?
We are currently not able to reproduce your issue.

Best,
Matthias

tiimjcb, 2024-11-16:
Hello.
AlDente is crashing every time I click on a button.

Here's the crash report

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               AlDente [7474]
Path:                  /Applications/AlDente.app/Contents/MacOS/AlDente
Identifier:            com.apphousekitchen.aldente-pro
Version:               1.28.6 (69)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-11-16 16:58:24.2692 +0100
OS Version:            macOS 15.1 (24B2082)
Report Version:        12
Anonymous UUID:        CA16B6F9-EE0B-90D1-3042-72D4B548D2CC

Sleep/Wake UUID:       062A538B-72EB-4EA9-A326-C9DD3B0E2E23

Time Awake Since Boot: 1800 seconds
Time Since Wake:       1241 seconds

System Integrity Protection: enabled

Crashed Thread:        4

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x0000000102c9572c

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [7474]

Thread 0::  Dispatch queue: com.apple.main-thread
0   AppKit                        	       0x18a9b7724 -[NSViewController view] + 0
1   SwiftUI                       	       0x1b5ac7bc8 static PlatformViewRepresentable<>.platformView(for:) + 24
2   SwiftUI                       	       0x1b5ac7ba8 protocol witness for static PlatformViewRepresentable.platformView(for:) in conformance PlatformViewRepresentableAdaptor<A> + 20
3   SwiftUI                       	       0x1b6308930 PlatformViewHost.representedView.getter + 124
4   SwiftUI                       	       0x1b62c5790 PlatformViewHost.focusView.getter + 24
5   SwiftUI                       	       0x1b5242140 NSViewResponder.focusRingView.getter + 224
6   SwiftUI                       	       0x1b55d4d40 FocusBridge.didChangeFocusItem(from:to:) + 1120
7   SwiftUI                       	       0x1b55d4594 FocusBridge.focusedItem.setter + 336
8   SwiftUI                       	       0x1b5eff470 FocusBridge.firstResponderDidChange(to:rootResponder:) + 1120
9   SwiftUI                       	       0x1b58fc6c0 NSHostingView.focusDidChange() + 276
10  SwiftUI                       	       0x1b634388c closure #1 in FirstResponderObserver.notifyResponderChain(_:notified:) + 112
11  SwiftUICore                   	       0x222c300c4 partial apply for thunk for @escaping @callee_guaranteed () -> () + 28
12  SwiftUICore                   	       0x222ee4ba0 static Update.dispatchActions() + 1236
13  SwiftUICore                   	       0x222ee4084 static Update.end() + 212
14  SwiftUI                       	       0x1b6343598 FirstResponderObserver.notifyAffectedRenderers(_:) + 1268
15  SwiftUI                       	       0x1b6342da0 FirstResponderObserver.observeValue(forKeyPath:of:change:context:) + 188
16  SwiftUI                       	       0x1b634306c @objc FirstResponderObserver.observeValue(forKeyPath:of:change:context:) + 244
17  Foundation                    	       0x187fda11c NSKeyValueNotifyObserver + 252
18  Foundation                    	       0x188086ef4 NSKeyValueDidChange + 388
19  Foundation                    	       0x188873e78 NSKeyValueDidChangeWithPerThreadPendingNotifications + 160
20  AppKit                        	       0x18aa5cee8 -[NSWindow _realMakeFirstResponder:] + 520
21  AlDente                       	       0x102bf01d4 0x102bec000 + 16852
22  AlDente                       	       0x102c229d0 0x102bec000 + 223696
23  Foundation                    	       0x18803cbec __NSFireTimer + 104
24  CoreFoundation                	       0x186e5f824 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 32
25  CoreFoundation                	       0x186e5f4c8 __CFRunLoopDoTimer + 1012
26  CoreFoundation                	       0x186e5efd8 __CFRunLoopDoTimers + 356
27  CoreFoundation                	       0x186e449c0 __CFRunLoopRun + 1856
28  CoreFoundation                	       0x186e43bc4 CFRunLoopRunSpecific + 588
29  HIToolbox                     	       0x1922b5f64 RunCurrentEventLoopInMode + 292
30  HIToolbox                     	       0x1922bbd54 ReceiveNextEventCommon + 636
[... 422 lines cut ...]
      "deploymentId" : 250000008
    },
    {
      "rolloutId" : "60da5e84ab0ca017dace9abf",
      "factorPackIds" : {

      },
      "deploymentId" : 250000007
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000004
    },
    {
      "treatmentId" : "993c66e1-bf83-41fd-a09f-0d8d5906d223",
      "experimentId" : "664fc3cd79a61a0b8b557292",
      "deploymentId" : 500000013
    }
  ]
}
}
```

MatthiasKerbl, 2024-11-19:
> Hello. AlDente is crashing every time I click on a button.
>
> Here's the crash report
>
> ```
> -------------------------------------
> Translated Report (Full Report Below) [...]

Hi @tiimjcb,

Which button did you click so that AlDente crashed?

Best,
Matthias

tiimjcb, 2024-11-19:
Hello @MatthiasKerbl,
This happened when I clicked the slider to adjust the battery charge.

It also happens on any button. I can right click the menu bar, and open the settings, but then, I can't click anywhere (menu or action button) without a crash. This produces the same error. 

According to ChatGPT, it seems it's a SwiftUI related issue.

idfq, 2024-11-20:
My app does not crash while I'm on the MacBook Pro M4, but Al Dente does not work at all as you're aware. If you need any help to resolve this, let me know

MatthiasKerbl, 2024-11-20:
> My app does not crash while I'm on the MacBook Pro M4, but Al Dente does not work at all as you're aware. If you need any help to resolve this, let me know

Hi @idfq,

What exactly does not work?
If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.
For us, the current version of AlDente works almost flawlessly on the new M4 MacBook.

Best,
Matthias

idfq, 2024-11-20:
> Hi @idfq,
>
> What exactly does not work? If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a [...]

Hi Matthias, sorry, while searching for a solution I just figured out that going to sleep charges to 100%. I assumed it was due to the new M4 not being compatible. Ignore what I said!

MatthiasKerbl, 2024-11-21:
Hi @tiimjcb,

We have just released AlDente 1.29, which includes a fix for this issue. To get AlDente 1.29 you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Best,
Matthias