AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1371: Sailing mode based on common misconception

Azirine, 2024-11-12:
Contrary to popular belief, micro-cycles are beneficial for the performance of the Li-ion cells. The paper below found that micro-cycles (<2% DOD) increase lifespan in the range of 31% to 50%.

https://www.sciencedirect.com/science/article/pii/S2352152X2201338X

Sailing mode can inadvertently speed up the degration of numerous MacBook batteries, despite users' best intentions.

Xu-adam, 2024-11-14:
Good point!

MatthiasKerbl, 2024-11-20:
Hi @Azirine,

Thanks for the hint. We will check it out and if necessary make changes.

Best,
Matthias

ksdev-pl, 2024-12-04:
I’m curious about the conclusions from this article:
	•	Should we enable sailing mode for a 1% range (e.g., from 69% to 70%)?
	•	Or is having it set to 1% effectively the same as disabling sailing mode? For instance, does AlDente start charging when the battery drops from 70% to 69%, or does it account for a 0.1% drop?