AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1378: MacBook Air M2 crashing

itsnandy, 2024-11-20:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Air M2
 - macOS Version (https://support.apple.com/en-us/HT201260): 14.7.1 (23H222)
 - Charger Model (Apple, Third Party, Wattage,...): Belkin Thunderbolt Dock
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Thunderbolt 4
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.28.6
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? Yes, though using Lunar to disable them
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
My laptop is randomly crashing. Sometimes when it's asleep or as I am using it. Been happening for maybe 1-2 weeks now /:

**To Reproduce**
General use unfortunately

**Additional context**
Here is the panic log: 

panic(cpu 6 caller 0xfffffe0024f3fea8): DCP PANIC - ASSERT!AppleDCPDPTXPowerController.cpp:538 No device added after powering on the rails. HPD=0 - dcpav(27)
ASSERT!AppleDCPDPTXPowerController.cpp:538 No device added after powering on the rails. HPD=0
RTKit: RTKit-2419.140.12.release - Client: local-ipad14dcp.RELEASE
!UUID: 5e11798b-7e3b-312b-ba79-1200badef791
Time: 0x00000129e4eb2dfe

Faulting task  27 Call Stack: 0x000000000001eb20 0x000000000001e510 0x000000000001e314 0x00000000000175d4 0x000000000005873c 0x00000000000fe334 0x000000000001f268 0x00000000000fe088 0x0000000000058634 0x00000000001085c0 0x00000000000fdcac 0x000000000001ef94 0x0000000000019e7c
RTKit Task List:
   name                    | pri     | stack use | status     | resource | warning
 0 rtk_background          | 007     |   824/2048  | SEMWAIT    | 0x87de70 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

 1 rtk_ep_work             | 057     |   920/2048  | SEMWAIT    | 0x87a640 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

 2 log_tx                  | 007     |   624/2048  | RUNNABLE   | 0 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

 3 log_flush               | 015     |   848/4096  | SEMWAIT    | 0x866490 | 
      0x00000000000193e0 0x000000000001a28c 0x0000000000024e50 0x0000000000019e7c

 4 power                   | 000     |  1760/65536 | RUNNABLE   | 0x637dd8 | 
      0x00000000000193e0 0x000000000001dd7c 0x000000000001da64 0x0000000000019e7c

 5 Terminator              | 015     |   416/2048  | SEMWAIT    | 0xffffffff41173f20 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

 6 main                    | 015     |  6544/40960 | SEMWAIT    | 0x84d130 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x000000000000cdb8 0x000000000000ecf0

 7 dcpexpert               | 015     |   240/16384 | SEMWAIT    | 0xffffffff41170620 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

 8 AFKMailboxEndpoint      | 015     |   704/16384 | SEMWAIT    | 0xffffffff4116bf20 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

 9 rtkitpmgr               | 015     |   240/16384 | SEMWAIT    | 0xffffffff41166dc0 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

10 iomfb_driver            | 016     | 11360/16384 | SEMWAIT    | 0xffffffff41162730 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

11 iomfb_low               | 009     |   240/16384 | SEMWAIT    | 0xffffffff41158410 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

12 iomfb_driver_low        | 008     |   240/16384 | SEMWAIT    | 0xffffffff41148e80 | 
      0x00000000000193e0 0x000000000001a28c 0x000000000001ef28 0x0000000000019e7c

13 iomfb_cdfd              | 014     |  3808/16384 | SEMWAIT    | 0xffffffff4112a4d0 | 
[... 790 lines cut ...]
com.apple.iokit.IOAVFamily	1.0.0
com.apple.iokit.IOHDCPFamily	1.0.0
com.apple.iokit.IOCECFamily	1
com.apple.iokit.IOAudio2Family	1.0
com.apple.driver.AppleIISController	340.1
com.apple.driver.AppleAudioClockLibs	340.8
com.apple.driver.FairPlayIOKit	71.10.0
com.apple.driver.AppleARMPlatform	1.0.2
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.IOReportFamily	47
com.apple.security.quarantine	4
com.apple.security.sandbox	300.0
com.apple.iokit.IOStorageFamily	2.1
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.iokit.CoreAnalyticsFamily	1
com.apple.security.AppleImage4	6.3.0
com.apple.kext.CoreTrust	1
com.apple.iokit.IOCryptoAcceleratorFamily	1.0.1
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.Compression	1.0
com.apple.kec.corecrypto	14.0

** Stackshot Succeeded ** Bytes Traced 551514 (Uncompressed 1346064) **

MatthiasKerbl, 2024-11-21:
Hi @itsnandy,

I am sorry to hear you are experiencing these crashes.

1. Do you also run into these crashes when AlDente is disabled?
2. Do you also run into these crashes when AlDente is enabled but you are not using the Belkin Thunderbolt dock?

Best,
Matthias

sl0n9ra, 2024-12-31:
Hello @itsnandy 
I faced the same problem, Macbook Air M2. 
Did you manage to find a solution?

eweinhold, 2025-01-04:
MacBook Air M2 - Aldente freezes and ist not responding if started again.
Top-Bar Icons shows System Spinner on mouseover.
Battery is not charging anymore.
Even if the App is killed, the battery doesn't charge anymore until a complete reboot.
Then I can start Al Dente - ONCE… if I quit it - it seems to guit - but the battery doesn't charge anymore until a new reboot.
starting al dente again without a reboot causes the app to freeze again.

MatthiasKerbl, 2025-01-08:
Hi @sl0n9ra and @eweinhold,

If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.
Also, please share the crash reports of AlDente. You can find the crash reports by opening the “Console” app and navigating to “Crash Reports” on the left side. Afterward, you will see a list of all crash reports. Now, please export all AlDente crash reports individually by right-clicking on them and saving them as a .ips file.

Best,
Matthias

eweinhold, 2025-01-08:
@MatthiasKerbl I send a debug file via e-mail (to support@) 
The app doesen't crash, so no crash report is created - it freezes.
No interaction is possible anymore so I can't export a debug file in the frozen state *doh 
Even if I kill AlDente via the activity monitor (stalled) the battery is not charging anymore and the Mac will run out of power despite the power is plugged in, and the Magsafe LED is green - MacOS battery status keeps on *not charging* 

A reboot fixed that charging issue. Running AlDente ONE and NOT trying to quit it results in normal App behavior and everything behaves like it should - unless I try to quit AlDente -> frozen again.

MatthiasKerbl, 2025-01-08:
> @MatthiasKerbl I send a debug file via e-mail (to support@) The app doesen't crash, so no crash report is created - it freezes. No interaction is possible anymore so I can't export a debug file in the frozen state *doh Even if I kill AlDente via the activity monitor (stalled) the battery is not charging anymore and the Mac will run out of power despite the power is plugged in, and the Magsafe LED is green - MacOS battery status keeps on _not charging_
> 
> A reboot fixed that charging issue. Running AlDente and NOT trying to quit it results in normal App behavior and everything behaves like it should - unless I try to quit AlDente -> frozen again.

Hi @eweinhold,

Thanks for the detailed feedback.

- Are you using your MacBook in Clamshell mode?
- Do you have Automatic Discharge enabled within AlDente?
- Do you have another app installed that manipulates your MacBook's sleeping behavior?

Joeao, 2025-01-13:
I've also experienced my Macbook Air M2 crashing (Not the app - the device) with the same Problem Details as mentioned in the original comment.

For me this has happened when the following has occurred:
- when closing the lid on my laptop
- Al Dente is running
- charging has paused due to the limit being hit

It doesn't happen consistently but has happened many times. This didn't occur before I installed Al Dente, I'm not using a USB-C dock. 

I've sent a debug file to support@apphousekitchen. I can't provide a crash report as the app isn't crashing.

If there's any more info that I can provide, please ask.

---

@eweinhold I believe what you're describing is a separate issue as this one describes the device crashing whilst Al Dente is running, not Al Dente itself.

closed 2025-04-15 by MatthiasKerbl