AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1382: After the shutting down the mac with the power button, it loses the license key and most configuration.

ivanhernandez-dev, 2024-11-21:
**Describe the bug**
After the shutting down the mac with the power button, it loses the license key and most configuration.

**To Reproduce**
Steps to reproduce the behavior:
1. Shutdown your Mac holding the power button
2. Turn it back on
3. Wait until aldente starts at login
4. Aldente has lost license and configuration

MatthiasKerbl, 2024-11-21:
Hi @ivanhernandez-dev,

I am sorry to hear you are facing issues with AlDente.
Which AlDente Version do you use? If it is not 1.29, the newest one, please install this one and let me know if you still run into this issue.

Best,
Matthias

ivanhernandez-dev, 2024-11-21:
Thank you for your prompt reply @MatthiasKerbl . I was using version 1.28.4, and after updating to 1.29, I haven’t been able to reproduce the issue.

You can go ahead and close this for now. If the problem occurs again, I’ll reopen it.

Apologies for any inconvenience caused.

closed 2024-11-22 by MatthiasKerbl