AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1383: Dutch translation enhancements

FrederikT123, 2024-11-22:
After the latest update (version 1.29) I noticed some small translation errors.
1. Current translates to "Stroom" instead of "Actueel"
2. Time to full translates to "Tijd tot opgeladen" instead of "Tijd naar volle"
3. You could translate "low power mode" as "energiebesparingsmodus"

These are the errors that I noticed but it is possible there are some more. Thanks in advance for fixing this minor issue :)

[image]

MatthiasKerbl, 2024-11-28:
Hi @FrederikT123,

Thanks for letting us know.
We will fix it with the next update.

Best,
Matthias

NoordWest, 2024-11-30:
In addition to the above mentioned corrections for the Dutch translation, I would like to add the following.
 Dutch translation in MacOS for 'sleep' is 'sluimer' as can be seen here
[image]

So to use consistent naming the following could be modified:
[image]

Translation of 'off" should be 'uit' instead of 'af'. 
[image]

closed 2025-05-15 by MatthiasKerbl

NoordWest, 2025-05-16:
> MatthiasKerbl
>  left a comment
> (AppHouseKitchen/AlDente-Charge-Limiter#1383)
> Hi @NoordWest <https://github.com/NoordWest>,
>
> We have just released AlDente 1.33, which includes a fix for this issue. > To get AlDente 1.33, you just need to update within the app under Settings/General/Check for Updates.
> Please report back if it works for you.
>
> Best,
> Matthias

Hi Mattias,

Almost correct, think you overlooked 1 item. Please see attached screenshot. 
My proposal for the modified text is written in red.

Other translations worked out fine.

best regards
Rudi

￼