AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1384: AllDente high disk write count (4GB/week)

smiba, 2024-11-22:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Pro M3 Max (16") 16-core
 - macOS Version (https://support.apple.com/en-us/HT201260): MacOS 15.1
 - Charger Model (Apple, Third Party, Wattage,...): 90W USB-C Dock (Dell Ultrasharp display)
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Third Party USB-C thunderbolt
 - AlDente Pro or AlDente Free: AlDente Free
 - AlDente Version: 1.28.6
 - List or add screenshots of all enabled/disabled settings in AlDente: It's the free version, so just the default stuff. I have my limit currently set to 68%
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/). 
[aldentepro-debug-2024-11-22-18:53:17.pasta.zip](https://github.com/user-attachments/files/17874286/aldentepro-debug-2024-11-22-18.53.17.pasta.zip)
 - External Monitor connected? Yes
 - MacBook used in Clamshell Mode (with the lid closed)? Yes

**Describe the bug**
AllDente seems to write about ~700MB to disk a day, resulting in currently 5.3GB written in 9 days. Although this is not the end of the world, it does seem like quite a lot for a tool I merely run to limit my battery charge. It would be nice if AlDente wouldn't run through a full SSD wear cycle every year. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open AlDente
2. Check in after a week or so

**Expected Behavior**
I expect this tool to write no more than 10MB/day, ideally even less? I don't understand what it could possibly be writing about really.

dinkopetkoff, 2024-11-27:
I can confirm the same issue on MacBook M1 Pro too. AlDente shows ~770MB in Bytes written in Activity Monitor (for about 8h of working on the Mac per day). Wondering if those are real writes or some buffer/OS interaction from the app that is visualized wrongly. In any case - we need confirmation (or a fix), since it is concerning for the SSD.

nthekim, 2024-11-28:
Exactly the same by me on M3 Macbook Pro 16 with Aldente Pro. Why need a small tool like that so much write operations on disk?

LIJiongyi, 2024-12-10:
I used it for 5 hours and it has about 70MB written to the disk. So about 51GB a year if we calculate...
Does it run a bit better on my Mac?(Macbook 14inch 2021)(macOS 15.1.1) I guess it's a problem with the frequency of calling the system API. Maybe the software can have an option to adjust the power monitoring frequency?

MatthiasKerbl, 2025-01-08:
Hi everyone,

Thank you for making us aware of this. We will look into it and try to improve it.
Since there is already a GitHub issue open for this (#1256), I will close this issue.

Best,
Matthias

closed 2025-01-08 by MatthiasKerbl