AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1385: Missing localized literals

andreuBiset, 2024-11-24:
**MacBook (please complete the following information):**
 - AlDente Pro or AlDente Free: **Pro**
 - AlDente Version: 1.28.4

**Describe the bug**
All labels and text are showing the key, not the localized text, maybe because I am using a not supported language (Catalan). 

**To Reproduce**
Steps to reproduce the behavior:
1. Open any modal or window
2. See the key of labels instead of localized text

**Expected Behavior**
If system language is not supported, use English as default language.

**Screenshots and Screen Recordings**

[image]
[image]
[image]

**Additional context**
I have AldentePro with SetApp https://setapp.com (Version 3.42.0)

MatthiasKerbl, 2024-11-24:
Hi @andreuBiset,

Thank you for notifying us about this issue. We will try to fix it as soon as possible.
For now, you should be able to fix it by following this [tutorial](https://apphousekitchen.com/step-by-step-guide-change-language-for-a-single-app-on-macos/).

Best,
Matthias

andreuBiset, 2024-11-24:
HI @MatthiasKerbl,

Thank you! After restarting, Setapp updated AlDente to version 1.29, and now the texts are displaying correctly:

[image]
[image]

closed 2024-11-24 by andreuBiset