AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1389: Serial number text overflows in Aldente stats popup

ksdev-pl, 2024-12-04:
Hello

**MacBook**

- AlDente Pro 1.29
- Macbook Pro M4
[image]
<br><br>

**Bug**

Serial number text overflows in Aldente stats popup:

[image]

MatthiasKerbl, 2025-04-19:
Hi @ksdev-pl,

Unfortunately, I am not able to reproduce this issue.
Do you still experience this issue?

Also, we have updated this feature. Now you can select each data point individually and hide the Serial Number, too.

Best,
Matthias

ksdev-pl, 2025-04-19:
Indeed, it's no longer the case in new version.