AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1390: My system version is macOS 12.2.1, running on an M1 chip, and currently, the installation of the auxiliary program is unresponsive.

AdminYlc, 2024-12-08:
Hello

MacBook

AlDente Pro 1.27
Macbook Air M1

[image]

Bug
The installation of the auxiliary program shows no response at all.

MatthiasKerbl, 2024-12-08:
Hi @AdminYlc,

Please follow our [uninstallation guide](https://apphousekitchen.com/how-to-uninstall-aldente-pro/).
Afterward, download and install AlDente again.

Best,
Matthias

AdminYlc, 2024-12-09:
This issue was resolved by completely uninstalling and reinstalling the program. However, the problem occurred suddenly without any warning and has happened twice. I hope the team can investigate the root cause.

MatthiasKerbl, 2024-12-09:
Hey @AdminYlc,

Good to hear that you could fix it. We will try to improve on this matter with future updates.

Best,
Matthias

closed 2024-12-09 by MatthiasKerbl