AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1393: App function stops working if I quit the app

TomHsiung, 2024-12-13:
**Is your feature request related to a problem? Please describe.**
Yes. I want to quit the AlDente app after I finish my configuration of the charge policies. I want the AlDente runs as daemon process on my MacBook Pro.

**Describe the solution you'd like**
Daemon process and no need to keep the app open.

**Additional context**
I like the features provided by this app. It's great. I need a damone. I appreciate the work the team spending on this project. Thanks.

MatthiasKerbl, 2025-04-08:
Hi @adamigen,

It is possible to have AlDente running completely in the background (without the menubar icon and the dock icon).
Go to AlDente Settings/Appearance:

- Make sure to disable "% Show Percentage".
- Switch "Icon Style" to "Don't Show".
- Disable "Show Dock Icon".

Afterward, no menubar icon and no dock icon are visible, and AlDente also does not show up in the cmd+tab menu.
The only way to access AlDente again is through the Launchpad.

Best,
Matthias

adamigen, 2025-04-08:
@MatthiasKerbl thank you for your reply.
So, removing the annoyance is behind the paywall.

MatthiasKerbl, 2025-04-08:
> [@MatthiasKerbl](https://github.com/MatthiasKerbl) thank you for your reply. So, removing the annoyance is behind the paywall.

Hi @adamigen,

Changing the menubar icon, including not showing it at all, is only available in AlDente Pro.

Best,
Matthias