AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1394: High Memory usage after running AlDente continuously for a month

A-Kun, 2024-12-14:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Air 13-inch, M3, 2024
 - macOS Version (https://support.apple.com/en-us/HT201260): 14.7.1 (23H222)
 - Charger Model (Apple, Third Party, Wattage,...): Apple 67W
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe 3
 - AlDente Pro or AlDente Free: AlDente Free
 - AlDente Version: 1.29
 - List or add screenshots of all enabled/disabled settings in AlDente: [image]
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
[aldentepro-debug-2024-12-14-18:15:42.txt](https://github.com/user-attachments/files/18138084/aldentepro-debug-2024-12-14-18.15.42.txt)
 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
A clear and concise description of what the bug is.
AlDente is using 1.16 GB of memory after leaving it running for a month. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open AlDente
2. Run it for a month

**Expected Behavior**
A clear and concise description of what you expected to happen.
AlDente should use a reasonable amount of memory even after running for a long time. 

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2025-01-08:
Hi everyone,

Thank you for making us aware of this. We will look into it and try to improve it.
Since a GitHub issue is already open for this (#1256), I will close this issue.

Best,
Matthias

e-kotov, 2025-01-08:
> Since a GitHub issue is already open for this (#1256), I will close this issue.

But it's not the same. Over there the issue is about disk writes, here it's about memory usage. (though these may be connected... @MatthiasKerbl will of course know better)

MatthiasKerbl, 2025-01-08:
> > Since a GitHub issue is already open for this (#1256), I will close this issue.
> 
> But it's not the same. Over there the issue is about disk writes, here it's about memory usage. (though these may be connected... @MatthiasKerbl will of course know better)

Hi @e-kotov,

You are right, my mistake.
So far, we have not been able to reproduce this issue. Do you know how to trigger it?

Best,
Matthias

e-kotov, 2025-01-08:
> So far, we have not been able to reproduce this issue. Do you know how to trigger it?

Not yet. I currently have 25 days of uptime of macOS, but I have restarted Al Dente on the day when I posted the comment here. Since then (it's been one week), Al Dente is at 186 Mb uncompressed memory. Which is not a small footprint, but not too high for me to be concerned.

If I notice high usage again, I will report. Do you have any idea of what you would need from me or anyone else who spots this in terms of technical details...? A memory dump/sample using the Activity Monitor..?

A-Kun, 2025-01-10:
I was able to reproduce this again after 24 days of uptime:
[image]

kbot983, 2025-05-16:
I have the same issue after 52 days of uptime

[image]

leeronr, 2025-06-27:
I've noticed this happen several times, though I haven't noticed if its incremental.

Uptime 23+ days. 1.07GB in memory. 

M2 Pro MacBook Pro, 16GB, macOS 15.5

Anywhere-Music-Player, 2025-07-04:
Same here 

[image]
[image]
[image]

liguiyuang, 2025-07-05:
我这里也是有同样的问题，内存每天都在增加

MatthiasKerbl, 2025-08-29:
Hi everyone,

We have just released AlDente 1.35, which includes a fix for this issue. To update to AlDente 1.35, open the app and navigate to Settings > Check for Updates.
Please report back if it works for you.

Best,
Matthias