AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1395: Updated to macOS 15.2 and it's not working properly aldente

DruChill, 2024-12-15:
- MacBook Model (https://support.apple.com/es-es/111883): MacBook Air M1
 - macOS Version (https://support.apple.com/en-us/109033): macOS Sequoia 15.2
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): USB-C by Apple
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.29

**Describe the bug**
Since I updated to macOS sequoia 15.2 it doesn't work correctly aldende it doesn't stop the power well and when I disconnect it it still has the paused loading icon there are many errors help please I'm pro version, the .paste file here:

- debug file (https://www.mediafire.com/file/g5f1utx3oievyoa/aldentepro-debug-2024-12-15-09-19-50.pasta/file)

DruChill, 2024-12-15:
Apparently I think my adapter is starting to fail and could be the cause of this problem. In any case, my .past file could be useful

DruChill, 2024-12-17:
Apparently so, but I think the other entry on the Mac has gone bad and I'm using the other one, it's still a mystery.

MatthiasKerbl, 2024-12-17:
Hi @DruChill,

MacOS 15.2 is working without any issues on all of our devices.
Going through your debug file, I see that the issue seems to be somewhere in the power delivery (charger, cable, plug).

Best,
Matthias

DruChill, 2024-12-17:
If you are absolutely right, it seems that the entry of my MacBook failed 😞

closed 2024-12-17 by DruChill