AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1399: Scheduled discharge task prevents Mac from waking up

TomHsiung, 2024-12-23:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): M1MAX
 - macOS Version (https://support.apple.com/en-us/HT201260): 15.2
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Apple
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.29
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? Yes
 - MacBook used in Clamshell Mode (with the lid closed)? Yes

**Describe the bug**
Sleep my mac first and keep the sleep status, then the schedule time reached and it triggers the discharge process (orange MagSafe light flickers). Then try to wake up mac but the external monitor is not responsive.

Have to pull off the MagSafe cable to wake up the Mac.

**Expected Behavior**
Waking up as usual

TomHsiung, 2024-12-23:
Oops! It might be nonsense to trigger discharge during the sleep status becuase in sleep status Mac discharges little. Hmmm, 🤔

TomHsiung, 2025-01-18:
Need a fix of this issue because sometimes I will go out after waking up and the MacBook is in sleep. The schedulde discharge event will happen when the MacBook is in sleep. And then I come back, I cannot wake up it.

MatthiasKerbl, 2025-07-15:
Hi @TomHsiung,

We have just released AlDente 1.34.1, which includes a fix for this issue. To get AlDente 1.34.1, you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias