AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1407: Energy Mode selection function to AlDente Pro

lainvoo, 2025-01-07:
**Is your feature request related to a problem? Please describe.**
The only reason I need to also keep the stock battery icon is to be able to quickly change the Energy Mode: "Automatic", "Low Power", "High Power". 

**Describe the solution you'd like**
Could this feature be added to AlDente Pro? It would be nice to have a  toggle under preferences to enable "Energy Mode" selection in the "Popover" menu. This way I could choose any of the three modes. You have probably some research into this as there is a Low Power mode toggle in AlDetne Preferences under Features, but in order to activate it the preference pane needs to be opened which is too many clicks.

mikicvi, 2025-01-20:
I am so down to see this on the Pro feature list! The only thing that is missing is a full replacement in the status bar!

vatsyy, 2025-04-25:
@lainvoo This should be extremely easy to implement given that we already have the CLI tool `pmset`.

sudo pmset -<b|c|a> lowpowermode <0|1|2> uses the -b, -c, or -a flag to specify battery, charger, or all power contexts whilst the trailing numeric argument sets the energy mode: 0 for Automatic, 1 for Low Power Mode, and 2 for High Power Mode (if supported by hardware).