AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1412: Notifications for calibration process

Razer0123, 2025-01-15:
It would be helpful to receive notifications for each target of the calibrations instead of having to check manually

It's a waste of time and mental resources to go check every X minutes the progress of calibration

Receiving notifications when the battery target level is reached is much more user friendly

devon386, 2025-01-21:
IMO this style of implementation would be annoying. Changing the icon symbol in the menubar would be better and less intrusive, plus you could see the calibration state at a quick glance

Razer0123, 2025-01-21:
> IMO this style of implementation would be annoying. Changing the icon symbol in the menubar would be better and less intrusive, plus you could see the calibration state at a quick glance

Would it?

Considering you have to calibrate every two weeks max and it's just 3 notifications

Changing the icon symbol would still mean you have to check yourself every other day, not really user friendly

devon386, 2025-01-22:
I see it as 5 notifications :  Calibration Warning - 100%- 15% - 100/1h - discharge/completion

I think you could easily adjust the battery icon to show symbology for the status of calibration. There are a lot of things that provide notifications, it's overkill. Al Dente should not add to that list.  It's not like you can speed up the calibration process by receiving a notification of its state.. For that reason, I believe status via menubar icon changing and that is adequate and provides enough information for the end user to make a decision.

Blarnix, 2025-03-22:
Giving the user a choice to either enable or disable those notifications couldn’t hurt, right?