AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1415: Low Power Mode Tint doesn't work in macOS Colored Icon Style

axinorm, 2025-01-22:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): Macbook Pro 14-inch, Nov 2024
 - macOS Version (https://support.apple.com/en-us/HT201260): macOS 15.2 (24C101)
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.30
 - List or add screenshots of all enabled/disabled settings in AlDente:

[image]

 - External Monitor connected? Yes
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
The battery icon is not orange like in other Icon Style modes.

**To Reproduce**
Steps to reproduce the behavior:
1. Select "macOS Colored" Icon Style in Appearance > Menubar Icon
2. Enable "Low Power Mode Tint"

**Expected Behavior**
After activating the "Low Power Mode Tint", the battery icon should be orange to distinguish it from normal mode.

MatthiasKerbl, 2025-01-28:
Hi @axinorm,

Thank you for your feedback. The Low Power Mode Tint is only applied to the battery frame. This is done due to the Low Power Mode Tint otherwise conflicting with the other colors of the macOS Colored Menubar icon style. I agree, that it is not visible enough and we will try to improve it.

Best,
Matthias

axinorm, 2025-01-29:
Hi @MatthiasKerbl,

Thanks a lot for your answer.

Regards.

MatthiasKerbl, 2025-04-19:
Hi @axinorm,

We have updated AlDente, and now it shows the LPM by a small orange dot next to the battery icon.

Best,
Matthias

closed 2025-04-19 by MatthiasKerbl