AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1417: "Disable Sleep until Charge Limit" unexpectedly disable display sleep even if charge limit is reached

dingwen07, 2025-01-24:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): Mac15,10
 - macOS Version (https://support.apple.com/en-us/HT201260): macOS 15.2 (24C101)
 - Charger Model (Apple, Third Party, Wattage,...): Third Party, 100W
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe 3
 - AlDente Pro or AlDente Free: AlDente Pro
 - AlDente Version: 1.30
 - List or add screenshots of all enabled/disabled settings in AlDente: Disable Sleep until Charge Limit
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? False
 - MacBook used in Clamshell Mode (with the lid closed)? False

**Describe the bug**
A clear and concise description of what the bug is.

```
   pid 21545(com.apphousekitchen.aldente-pro): [0x000008ee000592c4] 00:00:07 PreventUserIdleDisplaySleep named: "AlDente-Pro"  
   pid 21545(com.apphousekitchen.aldente-pro): [0x000008ee000792c5] 00:00:07 PreventSystemSleep named: "AlDente-Pro"  
```

**To Reproduce**
Steps to reproduce the behavior:
1. Enable "Disable Sleep until Charge Limit"
2. Charge to charge limit
3. Run `pmset -g assertions` and observe MacBook won't go to sleep

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

schulzj01, 2025-04-22:
I recently came across this issue as well.  I could not force Al Dente to remove it's lock on the Sleep function.  Even after disabling "Disable Sleep until Charge Limit",  the Sleep option in the  menu would not enable for me.  

Re-installing the Helper application from the Al Dente menu did reset this for me temporarily, however shortly afterwards, Sleep was disabled again.  Now the only way I'm able to sleep OSX is to quit Al Dente

MatthiasKerbl, 2025-04-23:
Hi @dingwen07 and @schulzj01,

The following blog post describes all scenarios in which AlDente has to disable sleep: "[When Does AlDente Disable Sleep on MacBooks?](https://apphousekitchen.com/when-does-aldente-disable-sleep-on-macbooks/)"

If AlDente is still disabling Sleep outside of these scenarios, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

schulzj01, 2025-04-23:
Thank you @MatthiasKerbl.  I wasn't aware of that blog post.  In my case, it appeared as though the issue was that I had the "Automatic Discharge" enabled and my charge level was slightly above the limit set which disabled Sleep. Thank you for the clarification on this!

closed 2025-05-09 by MatthiasKerbl