AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1426: I AM UNABLE TO INSTALL HELPER. I am unable to Install helper I have tried so many times But it shows install helper error and asks for password again and again. I've tried all the pr

BERSERK555, 2025-02-08:
**MacBook (please complete the following information):**
 - MacBook Model ( macbook air m1 2020):
 - macOS Version ( Sonoma 14.2.1):
 - Charger Model (originalapple charger):
 - Charging Cable ( original cable):
 -  AlDente Free:
 - AlDente Version: latest 
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected no
  
**Describe the bug**
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2025-02-10:
Hi @BERSERK555,

I am sorry to hear you are having issues with AlDente. Issues like that might occur when multiple versions of AlDente are installed. Therefore, please follow this guide [How to uninstall AlDente Pro](https://apphousekitchen.com/how-to-uninstall-aldente-pro/). Afterward, please check if you have any other AlDente anywhere else on your MacBook (Downloads folder, Application folder,...). Lastly, download and install the newest version of AlDente.

Best,
Matthias

simon-bd, 2025-02-18:
i face a similar issue currently this is what happens after a clean install following the guide and removing all files when installing : 

[image]

[image]

[image]

after i press OK, nothing happend and i cant open the UI of Aldente

simon-bd, 2025-02-19:
I managed to resolve my issue by ensuring that the launchdemons and launchagents had the correct ownership and permissions. By the way, @BERSERK555, could you please check the permissions for the following directories: `ls -ld ~/Library/LaunchAgents /Library/LaunchAgents /Library/LaunchDaemons`? It might be the same issue, but I think this is a good test to confirm.

MatthiasKerbl, 2025-02-20:
> I managed to resolve my issue by ensuring that the launchdemons and launchagents had the correct ownership and permissions. By the way, [@BERSERK555](https://github.com/BERSERK555), could you please [...]

Hi @simon-bd,

Thank you for your update. Do you have multiple user accounts on your MacBook?

Best,
Matthias

simon-bd, 2025-02-21:
@MatthiasKerbl I'm pretty sure I did it my self I was working on a custom launch agent for displaylink manager and had some permission issue in terminal. 

I only have one user

closed 2025-04-15 by MatthiasKerbl