AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1429: Chinese Translation and Formatting Issue in Power Adapter Specification

ktywb, 2025-02-11:
I am a user of this product, and I recently noticed a translation and formatting issue in the power adapter specification section. The current display appears to be unclear and may cause misunderstanding.

The issue is as follows:

- Current format:
```
当前： 0.4 A 的 2.0 A  
电压  19.8 V 的 20 V  
动力  7.7 W 的 40 W  
```
- > The use of “的” in this context is ambiguous. It is unclear whether it indicates a range or a comparison. The phrase "0.4 A 的 2.0 A" seems to be a mistranslation and is confusing. A direct English translation would be "0.4 A of 2.0 A", which does not make sense in this context. 
- > The term “动力” (meaning "power" in the sense of mechanical force) is not appropriate for describing electrical power. “功率” should be used instead.

I hope this feedback is helpful, and I would appreciate it if you could correct this issue in a future update.

MatthiasKerbl, 2025-02-11:
Hi @ktywb,

Thanks for letting us know about these translation mistakes. We will fix them with the next update.

Best,
Matthias

MatthiasKerbl, 2025-03-06:
Hi everyone,

We have just released AlDente 1.31.3, which includes a fix for this issue. To get AlDente 1.31.1, you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Best,
Matthias

closed 2025-03-06 by MatthiasKerbl

ktywb, 2025-03-06:
Thank you for the update! It looks much better now! I really appreciate the quick fix.