AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1435: Update Signing Warning/Error (1.30 -> 1.31)

TrevorMahlmann, 2025-02-20:
**MacBook (please complete the following information):** Air
 - MacBook Model (https://support.apple.com/en-us/HT201608): M2
 - macOS Version (https://support.apple.com/en-us/HT201260): 15.3.1
 - Charger Model (Apple, Third Party, Wattage,...): N/A
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): N/A
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: Updating from 1.30 to 1.31
 - List or add screenshots of all enabled/disabled settings in AlDente: Attached below
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
When updating, a warning appears saying the update is not properly signed.

**To Reproduce**
Steps to reproduce the behavior:
Be on 1.30, update to 1.31.

**Expected Behavior**
Update to 1.31

**Screenshots and Screen Recordings**
Attached below

**Additional context**

[aldentepro-debug-2025-02-20-10:36:34.pasta.zip](https://github.com/user-attachments/files/18889312/aldentepro-debug-2025-02-20-10.36.34.pasta.zip)

[image]

aletei3, 2025-02-20:
I second this.
2021 MacBook Pro M1 Pro 
MacOS 15.3.1

ljara2106, 2025-02-20:
Getting same error on M4 Pro

kjoonlee, 2025-02-21:
Download starts and continues until around 700 KB
Then it restarts the download
Download starts and continues until around 13 MB
Then the error comes up.

I’m running macOS 15.3.1 as well.

StormyGalaxy, 2025-02-21:
I have the same issue on my Macbook Air M2, Running macOS 15.2 Beta

Real-YueWu, 2025-02-21:
Same error on my Macbook Air M2 with macOS version 14.7.4

bmbravo, 2025-02-21:
Same error here, m3 max macbook with 15.3.1 macOS

MatthiasKerbl, 2025-02-21:
Hi everyone,

We have made some changes and the update should work now as expected.
Apologies for the inconvenience and delay.

Best,
Matthias

aletei3, 2025-02-21:
Thank you for the improvements made in this update and for the instant support you devs always give us. 
I wanted to point out there's a minor visual bug with how the "native iOS" icon is rendered on the menu bar (btw, I love it, it shows all the necessary information without occupying too much space). 
See attachment.

Can't restart my machine at the moment, updates will follow.

2021 MacBook Pro M1 Pro
AlDente Pro 1.31

[image]

MatthiasKerbl, 2025-02-24:
> Thank you for the improvements made in this update and for the instant support you devs always give us. I wanted to point out there's a minor visual bug with how the "native iOS" icon is rendered on the menu bar (btw, I love it, it shows all the necessary information without occupying too much space). See attachment.
> 
> Can't restart my machine at the moment, updates will follow.
> 
> 2021 MacBook Pro M1 Pro AlDente Pro 1.31
> 
> <img alt="Image" width="48" src="https://private-user-images.githubusercontent.com/160548462/415555806-acdf765d-9328-4208-ad65-47047cc05d04.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDAzNjYzNDQsIm5iZiI6MTc0MDM2NjA0NCwicGF0aCI6Ii8xNjA1NDg0NjIvNDE1NTU1ODA2LWFjZGY3NjVkLTkzMjgtNDIwOC1hZDY1LTQ3MDQ3Y2MwNWQwNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIyNFQwMzAwNDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NDc4YzI0NzE0MTg1YzE2OGUyYTc1MTJkM2IwMTliZDAyYzFkM2UzZDRkMjg0NDc1MTNmNGU5Y2ZiMmI5YmU1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.jFDvSovf2fqmvJiVPEnzH4wwRa7ptr33LIGulh5Wdl4">

Hi @aletei3,

Thank you for the positive feedback, that's much appreciated.
Also, thank you for letting us know about the defect in the icon. We will look into it.

Best,
Matthias

closed 2025-02-24 by MatthiasKerbl