AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1436: Update to 1.31 Ditaline

petervannes, 2025-02-20:
**MacBook (please complete the following information):**
 - MacBook Model MK193N/A
 - macOS Version 15.3.1 (24D70)
 - Charger Model Third Party LG Display
 - Charging Cable Thunderbold 2
 - AlDente Pro or AlDente Free:Pro
 - AlDente Version: 1.30
 

**Describe the bug**
Update of All Dente Pro from 1.30 to 1.31 fails

**To Reproduce**
Steps to reproduce the behavior:
1. Open Settings > General 
2. Select Check for updates
3. Software update window pops up
4. Select Install Update
5. Progress shown
6. - Downloading Update
7. - Extracting update
8. Error window pops up after about 3,8MB download
9. " Update Error! An error occured while extracting the archive. Please try again later." 
10. 

https://github.com/user-attachments/assets/28d222e6-4f00-4c1a-a0db-47bb3247045b

MatthiasKerbl, 2025-02-21:
Hi everyone,

We have made some changes and the update should work now as expected.
Apologies for the inconvenience and delay.

Best,
Matthias

petervannes, 2025-02-21:
👍🏻 Works as expected now, thanks!

closed 2025-02-22 by MatthiasKerbl